libuvc/libuvc #180: Build test failed

deeprun, 2021-05-25:
When I build with the test program with opencv 4.1, I got the following error

[ 87%] Building C object CMakeFiles/uvc_test.dir/src/test.c.o
In file included from /usr/include/opencv4/opencv2/core/types_c.h:82:0,
                 from /usr/include/opencv4/opencv2/core/core_c.h:48,
                 from /usr/include/opencv4/opencv2/highgui/highgui_c.h:45,
                 from /home/nvidia/apps/libuvc/src/test.c:35:
/usr/include/opencv4/opencv2/core/cvdef.h:690:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"
 #  error "OpenCV 4.x+ requires enabled C++11 support"
    ^~~~~
/usr/include/opencv4/opencv2/core/cvdef.h:695:10: fatal error: array: No such file or directory
 #include <array>
          ^~~~~~~
compilation terminated.
CMakeFiles/uvc_test.dir/build.make:62: recipe for target 'CMakeFiles/uvc_test.dir/src/test.c.o' failed
make[2]: *** [CMakeFiles/uvc_test.dir/src/test.c.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/uvc_test.dir/all' failed
make[1]: *** [CMakeFiles/uvc_test.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

I added the following lines to CMakeLists.txt, but the problem still there. 

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)

How can I fix this?

goldie-lin, 2021-05-26:
Try to add:
```
  target_compile_options(uvc_test
    PRIVATE $<$<COMPILE_LANGUAGE:C,CXX>:-std=c++11 -x c++>
  )
```
after:
```
  add_executable(uvc_test src/test.c)
```
in `CMakeLists.txt` file.

deeprun, 2021-05-26:
@goldie-lin Thank you very much. 

After I made the changes, I got the following error:

CMake Error at CMakeLists.txt:151 (target_compile_options):
  Error evaluating generator expression:

    $<COMPILE_LANGUAGE:C,CXX>

  $<COMPILE_LANGUAGE> expression requires one or zero parameters.

How can I fix it?

goldie-lin, 2021-05-26:
What's your `cmake --version`?

deeprun, 2021-05-26:
3.10.2

goldie-lin, 2021-05-26:
It seems that that expression is too new fashion and unnecessary complex and not portable, then try this updated one:
```
  target_compile_options(uvc_test
    PRIVATE -std=c++11 -x c++
  )
```

chenjinlei, 2021-07-02:
@goldie-lin It helps, thank you.

mcuee, 2026-09-06:
#307 has been merged and this should be sorted out.

closed 2026-09-06 by mcuee