libuvc/libuvc #182: Invalid initial frame or two on start streaming (PureThermal 2)

mrspirytus, 2021-06-08:
I noticed that my application at times after calling `uvc_start_streaming(_devHandle, &_ctrl, frame_callback, this, 0);` I get one of two frames that are invalid. This does not happen all the time. But when it happens it only happens for one or two consecutive first frames. This happens with the latest commit and I sampled a few all the way to last summer 2020. Not sure if this is an issue with [PureThermal 2](https://groupgets.com/manufacturers/getlab/products/purethermal-2-flir-lepton-smart-i-o-module) or `libuvc` or even `libusb`. I will attempt to debug this but any early pointers welcomed (code area/functions) :)

```
void frame_callback(uvc_frame_t* frame, void* opaque)
{
    // Gray scale 16bit images
    if(frame->data_bytes != (2 * frame->width * frame->height)) {
        if(frame->data_bytes == 0)
            std::cout << "frame: zero" << std::endl;
        else
            std::cout << "frame: expecting " <<
                            std::to_string(2 * frame->width * frame->height) <<
                            " got " << std::to_string(frame->data_bytes) <<
                            std::endl;
        return;
    }

    // All good here
    cv::Mat img(frame->height, frame->width,
            CV_16UC1,
            reinterpret_cast<uchar*>(frame->data));
}
```

JoelStienlet, 2021-07-19:
Hello,
I think that the segfault I've seen may be related to your bug, can you please try my PR to see if it solves your problem as well?
[PR for issue 183](https://github.com/libuvc/libuvc/pull/184)

mrspirytus, 2021-07-19:
I will in about 2 weeks. I report my findings