libuvc/libuvc #183: Segmentation fault

JoelStienlet, 2021-07-11:
Hello all,
First thank you for this library.
I'd like to report a segmentation fault happening with usb microscopes 1871:7670.
I think these microscopes send corrupted frames sometimes...
(note that I switched to libuvc because these microscopes crashed the linux kernel ! It's probably the same kind of error.)
I found that the segmentation fault occurs in frame.c in uvc_uyvy2bgr(), and more precisely in while (pbgr < pbgr_end) {...}.
What happens is that the buffer sizes don't match.
I corrected the error by replacing the test condition:
while(((pbgr + 3*8) <= pbgr_end) && ((pyuv + 4*8) <= pyuv_end)){ .... }
this way, a segfault can no longer occur, neither in the input nor in the output buffer.
I suppose the other conversion functions could be modified to avoid segfaults with other models as well.

goldie-lin, 2021-07-11:
Do you mean the `data_bytes` of  received `uvc_frame_t` was sometimes larger then actual size in corrupted frame?

JoelStienlet, 2021-07-11:
Yes this is exactly what I think is happening. There are 3 parameters: in->width, in->height and in->data_bytes. Normally in->data_bytes should match (width X height), but the corrupted frame has a smaller buffer than what would be required for an image of the given size. So when I add this test on the real input buffer size, then the segfault disappears.

goldie-lin, 2021-07-11:
It's better to put your error detection and workaround code of frame info correction in [`_uvc_populate_frame()`](https://github.com/libuvc/libuvc/blob/master/src/stream.c#L1320-L1377) of `stream.c`. It's called before each user frame callback functions. Fixes frame size there to prevent your frame callback function get the incorrect size.

You could search a keyword `is_isight`, and add a flag in the [`struct uvc_device_handle`](https://github.com/libuvc/libuvc/blob/master/include/libuvc/libuvc_internal.h#L270-L289), and follow the similar way to detect your device VIP:PID in [`uvc_open_internal()`](https://github.com/libuvc/libuvc/blob/master/src/device.c#L337-L364) of `device.c` and apply your workaround code only for your device.

JoelStienlet, 2021-07-11:
Thank you very much for your indications, I'll work on a proper fix for this camera. 
Can I submit a PR once finished so that it gets fixed upstream, or do you prefer to keep these workarounds outside of the library?

goldie-lin, 2021-07-11:
Ha, I'm not the maintainer or collaborator of libuvc, I have no idea on this, you could go ahead.
I just done similar work before but for some reason I kept the patch in private repository.

r00t-, 2022-03-09:
i made a fix for a similar issue: https://github.com/libuvc/libuvc/pull/211/files
(the fix in #209 did not fix the crashes i was seeing.)

JoelStienlet, 2022-03-09:
> i made a fix for a similar issue: #211 (the fix in #209 did not fix the crashes i was seeing.)

very possibly the same origin: the data buffer beeing too small. Have you also tried my patch [https://github.com/libuvc/libuvc/pull/184](https://github.com/libuvc/libuvc/pull/184) ?

JoelStienlet, 2022-03-11:
@r00t- yes, I agree that the memory accesses inside `uvc_yuyv2bgr()` require safety checks.
But the `memcopy()` in `_uvc_populate_frame()` also needs to be safe. The code needs patches at multiple places IMO.

drinkcat, 2026-11-14:
Same remark as #184, thanks!

closed 2026-11-14 by drinkcat