libuvc/libuvc #185: Compiling on windows, Libusb error.

CarlosCoSonicat, 2021-07-20:
Hi, I'm trying to compile libuvc for windows 10, I have downloaded the Libusb dll and includes and have them in system variable.

The problem is that when compiling cmake throws an error in add_library  "Target " uvc" links to target "LibUsb::LIBUSB" but the target was not found..... "

I'm using mingw 32 if that helps.

mcuee, 2021-08-06:
Try this patch: https://github.com/microsoft/vcpkg/blob/master/ports/libuvc/build_fix.patch

mcuee, 2021-08-06:
But it is easier to use the following fork. You can use the official libusb and not the fork mentioned in the readme. 
https://github.com/pupil-labs/libuvc

mcuee, 2022-05-08:
Latest build of https://github.com/pupil-labs/libuvc git head with libusb-1.0.26 (dynamic link) and pthreads-w32-2-9-1-release, using VS2019 (both 32bit and 64bit). 
[pupil_labs_libuvc.zip](https://github.com/libuvc/libuvc/files/8646865/pupil_labs_libuvc.zip)

FrostKiwi, 2022-12-06:
> The problem is that when compiling cmake throws an error in add_library "Target " uvc" links to target "LibUsb::LIBUSB" but the target was not found..... "
> 
> I'm using mingw 32 if that helps.

Had the same experience and solved it with a bit hackery, since I required libUVC specifiaclly built by MinGW64 and the linked https://github.com/pupil-labs/libuvc didn't support NV12, that I needed for my use-cae. Steps to build with MinGW64 and Windows 10 covered here: https://github.com/libuvc/libuvc/issues/245