libuvc/libuvc #188: libuvc 支持macos 11.5

seaCheng, 2021-07-26:
long time without any update,this great library don't even available run well on the latest macos 11.5,not even support macos 12,so who will solve the problem?

mcuee, 2021-08-06:
What is the problem? You need to provide the details.

mcuee, 2021-08-06:
As for macOS 12, that is under beta. There is an issue reported here under libusb project.
https://github.com/libusb/libusb/issues/972

mcuee, 2021-08-06:
Just tested it our under my Mac Mini M1 (macOS 11.5 Big Sur) using homebrew (built from source) and it seems to work fine.
```
			libuvcConfigVersion.cmake
libuvc/buildm1 on  master [?] via △ v3.21.1 ❯ ./example
UVC initialized
Device found
Device opened
DEVICE CONFIGURATION (0ac8:3420/[none]) ---
Status: idle
VideoControl:
	bcdUVC: 0x0100
VideoStreaming(1):
	bEndpointAddress: 130
	Formats:
	UncompressedFormat(1)
		  bits per pixel: 16
		  GUID: 5955593200001000800000aa00389b71 (YUY2)
		  default frame: 1
		  aspect ratio: 0x0
		  interlace flags: 00
		  copy protect: 00
			FrameDescriptor(1)
			  capabilities: 00
			  size: 640x480
			  bit rate: 768000-196608000
			  max frame size: 614400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(2)
			  capabilities: 00
			  size: 352x288
			  bit rate: 768000-196608000
			  max frame size: 202752
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(3)
			  capabilities: 00
			  size: 320x240
			  bit rate: 768000-196608000
			  max frame size: 153600
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(4)
			  capabilities: 00
			  size: 176x144
			  bit rate: 768000-196608000
			  max frame size: 50688
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(5)
			  capabilities: 00
			  size: 160x120
			  bit rate: 768000-196608000
			  max frame size: 38400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			StillFrameDescriptor
			  bEndPointAddress: 00
			  wWidth(1) = 640
			  wHeight(1) = 480
			  wWidth(2) = 352
			  wHeight(2) = 288
			  wWidth(3) = 320
			  wHeight(3) = 240
			  wWidth(4) = 176
			  wHeight(4) = 144
			  wWidth(5) = 160
			  wHeight(5) = 120
END DEVICE CONFIGURATION

First format: (YUY2) 640x480 30fps
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 2688
bInterfaceNumber: 1
Streaming...
Enabling auto exposure ...
 ... full AE not supported, trying aperture priority mode
 ... enabled aperture priority auto exposure mode
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
...
```

seaCheng, 2021-08-07:
thans for the reponse ,i am really sorry,that i didn't descrbe clear the detail for the issure last time, the really problem i come across is the that , i use the libuvc library to control a wencam in mac computer.but one of mine colleage's computer update to the macos12 which is beta version,that the libuvc can't work well ,the wrong code is "Access denied"

mcuee, 2021-08-07:
As for macOS 12, that is under beta. There is an issue reported here under libusb project. You may have to contact Apple for support.
https://github.com/libusb/libusb/issues/972

mcuee, 2021-08-07:
@seaCheng Please also edit your title, others may not understand Chinese. Your English is pretty okay and no need to use Chinese in the title.

mcuee, 2021-08-08:
As per https://github.com/libusb/libusb/issues/972 you can use latest git version of libusb and sudo to use libuvc.

The proper way is for the developer to get the right entitlement as per the following discussion: https://github.com/libusb/libusb/pull/911
> kUSBReEnumerateCaptureDeviceMask requires either the application to be running as root OR the application has the entitlement com.apple.vm.device-access AND the user has accepted the authorization for the capture with IOServiceAuthorize.

mcuee, 2021-08-08:
I have closed https://github.com/libusb/libusb/issues/972 as this is not an issue with libusb. Either Apple changes its way or you have to get the right entitlement as a developer.

mcuee, 2021-10-30:
https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device-and-claim-exclusive-access

Updates on Oct 2021: with the merging of pull request #911, support for detaching kernel drivers with authorization has been added. This will be included in the upcoming 1.0.25 release. Take note the underlying macOS capture APIs only work on the whole device and not on individual interfaces. So this will force all the kernel extensions (drivers) bound to all the interfaces of a USB Composite devices to be released (Issue #920). You will need to get the entitlement from Apple Developer support, create a provisioning profile with that entitlement, and build your app with that profile. Please take note that command line apps cannot use provisioning profiles and therefore cannot hold this entitlement.

mcuee, 2021-10-30:
More discussions here.
https://github.com/libusb/libusb/issues/1014

mcuee, 2021-11-04:
Based on feedbacks from the libusb discussions, libuvc is not the right library to use under macOS.
https://github.com/libusb/libusb/issues/1014

mcuee, 2021-11-05:
If you are just testing, then you can use latest libusb git and then sudo.

```

mcuee@mcuees-Mac-mini build % sudo ./example 
UVC initialized
Device found
Device opened
DEVICE CONFIGURATION (0ac8:3420/[none]) ---
Status: idle
VideoControl:
	bcdUVC: 0x0100
VideoStreaming(1):
	bEndpointAddress: 130
	Formats:
	UncompressedFormat(1)
		  bits per pixel: 16
		  GUID: 5955593200001000800000aa00389b71 (YUY2)
		  default frame: 1
		  aspect ratio: 0x0
		  interlace flags: 00
		  copy protect: 00
			FrameDescriptor(1)
			  capabilities: 00
			  size: 640x480
			  bit rate: 768000-196608000
			  max frame size: 614400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(2)
			  capabilities: 00
			  size: 352x288
			  bit rate: 768000-196608000
			  max frame size: 202752
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(3)
			  capabilities: 00
			  size: 320x240
			  bit rate: 768000-196608000
			  max frame size: 153600
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(4)
			  capabilities: 00
			  size: 176x144
			  bit rate: 768000-196608000
			  max frame size: 50688
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			FrameDescriptor(5)
			  capabilities: 00
			  size: 160x120
			  bit rate: 768000-196608000
			  max frame size: 38400
			  default interval: 1/30
			  interval[0]: 1/30
			  interval[1]: 1/14
			StillFrameDescriptor
			  bEndPointAddress: 00
			  wWidth(1) = 640
			  wHeight(1) = 480
			  wWidth(2) = 352
			  wHeight(2) = 288
			  wWidth(3) = 320
			  wHeight(3) = 240
			  wWidth(4) = 176
			  wHeight(4) = 144
			  wWidth(5) = 160
			  wHeight(5) = 120
END DEVICE CONFIGURATION

First format: (YUY2) 640x480 30fps
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 2688
bInterfaceNumber: 1
Streaming...
Enabling auto exposure ...
 ... full AE not supported, trying aperture priority mode
 ... enabled aperture priority auto exposure mode
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
...

```

mcuee, 2022-09-05:
Workaround which may not be easy to implement (using a codeless DriverKit Extension).
* https://github.com/libusb/libusb/issues/972#issuecomment-1236512964