libuvc/libuvc #189: Can't open Realsense 435 due to UVC 1.5 issue

Timmmm, 2021-08-08:
I have a Realsense 435 depth camera which supports UVC 1.5. I found an old issue here about UVC 1.5 support that has been closed and it seems like you guys think this library supports UVC 1.5? However if I try to open the Realsense camera it fails in [this code](https://github.com/libuvc/libuvc/blob/master/src/device.c#L1085)

```
  switch (info->ctrl_if.bcdUVC) {
  case 0x0100:
    info->ctrl_if.dwClockFrequency = DW_TO_INT(block + 7);
    break;
  case 0x010a:
    info->ctrl_if.dwClockFrequency = DW_TO_INT(block + 7);
    break;
  case 0x0110:
    break;
  default:
    UVC_EXIT(UVC_ERROR_NOT_SUPPORTED);
    return UVC_ERROR_NOT_SUPPORTED;
  }
```

because the value of `info->ctrl_if.bcdUVC` is `0x0150` (i.e. UVC 1.5).

If I patch it to add `case 0x0150: break;` it gets a bit further. Does this library actually support UVC 1.5?

It may be the fault of the Realsense 435 firmware because Mac's native UVC drivers also fail to open it, however it does work on Windows. And I'm pretty sure that `switch` should include `0x0150` either way.

jschlatow, 2021-08-23:
@Timmmm, I recently had trouble using an UVC 1.5 device as well and created a pull request (#187) with the changes that made it work for me. Would you like to give my PR a try?

mcuee, 2026-09-06:
PR #187 has been merged.

Close this for now. Please reopen if it does not work.

closed 2026-09-06 by mcuee