libuvc/libuvc #192: "Zombie" Handle If Example Exits Without Releasing Handle

donn, 2021-09-08:
OS: `Linux [redacted] 5.4.143-1-MANJARO #1 SMP PREEMPT Thu Aug 26 23:41:07 UTC 2021 x86_64 GNU/Linux`

Libraries:
* libusb - 1.0.24-2
* libuvc - 0.0.6-1

---

I'm using a UVC device that works fine with the example file, but the problem is if there is any anomaly in the app quitting where it fails to release certain resources, the computer has to be restarted before it can hook up to the device again.

(Note: I'm letting libuvc manage the libusb context.)

Unplugging and re-plugging does nothing. It's almost like there's a zombie somewhere keeping a handle on the device, except the app has been long killed. I'm not really sure what to kill here to release this handle. I was under the impression libusb and libuvc ran entirely in userspace...

donn, 2021-09-09:
Adding the requisite permissions to open the relevant usb devices without sudo seems to alleviate this issue, plugging and replugging just works now. Still not sure why a root process would have a zombie handle somewhere but not running as root is a better idea anyway

JWhitleyWork, 2022-01-31:
I am seeing this behavior too. However, it only happens if I disconnect from a device and then try to connect to it again. When I disconnect from the device, I get the following in my kernel log:

```
Jan 31 17:44:26 pop-os kernel: [  213.967581] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 6 comp_code 13
Jan 31 17:44:26 pop-os kernel: [  213.967588] xhci_hcd 0000:01:00.0: Looking for event-dma 00000000fed33d80 trb-start 00000000fed33c30 trb-end 00000000fed33d60 seg-start 00000000fed33000 seg-end 00000000fed33ff0
Jan 31 17:44:27 pop-os kernel: [  215.143387] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:44:28 pop-os kernel: [  216.166213] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:44:30 pop-os kernel: [  218.183368] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
```

After this, nothing appears to happen and everything is fine until I try to connect to the device again, in which case I get the following reset attempts:

```
Jan 31 17:45:28 pop-os kernel: [  275.436507] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
Jan 31 17:45:29 pop-os kernel: [  277.224040] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:30 pop-os kernel: [  278.247785] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:32 pop-os kernel: [  280.263744] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:33 pop-os kernel: [  280.584118] usb 2-2: device descriptor read/8, error -110
Jan 31 17:45:33 pop-os kernel: [  280.690573] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
Jan 31 17:45:33 pop-os kernel: [  281.287747] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:36 pop-os kernel: [  283.303894] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:37 pop-os kernel: [  284.327236] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:38 pop-os kernel: [  285.962953] usb 2-2: device descriptor read/8, error -110
Jan 31 17:45:38 pop-os kernel: [  286.277788] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
Jan 31 17:45:39 pop-os kernel: [  286.343020] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:40 pop-os kernel: [  287.367942] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:42 pop-os kernel: [  289.383869] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:43 pop-os kernel: [  290.407438] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:44 pop-os kernel: [  291.335024] usb 2-2: device descriptor read/8, error -110
Jan 31 17:45:44 pop-os kernel: [  291.442763] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
Jan 31 17:45:45 pop-os kernel: [  292.423384] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:46 pop-os kernel: [  293.446879] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:48 pop-os kernel: [  295.462924] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:49 pop-os kernel: [  296.487956] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:45:49 pop-os kernel: [  296.715104] usb 2-2: device descriptor read/8, error -110
Jan 31 17:45:49 pop-os kernel: [  297.028873] usb 2-2: reset SuperSpeed USB device number 2 using xhci_hcd
```

and then the following forever more until I *hard* reboot:

```
Jan 31 17:47:22 pop-os kernel: [  389.704375] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:23 pop-os kernel: [  390.728261] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:25 pop-os kernel: [  392.744301] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:26 pop-os kernel: [  393.768320] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:28 pop-os kernel: [  395.784359] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:29 pop-os kernel: [  396.812341] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:31 pop-os kernel: [  398.824402] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:32 pop-os kernel: [  399.848417] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:34 pop-os kernel: [  401.864565] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:35 pop-os kernel: [  402.888441] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
Jan 31 17:47:37 pop-os kernel: [  404.904459] usb 1-5: io_dev_set_duty failed: 110: io_dev_write
```

JWhitleyWork, 2022-01-31:
Looks like the `io_dev_set_duty failed` are just a side-effect which are specific to the `system76-io` package.

donn, 2022-02-01:
Pop!_OS or not, the behavior you're describing is identical to what I faced.