libuvc/libuvc #193: SIGFPE if endpoint_bytes_per_packet == 0

dmikushin, 2021-09-11:
There is potentially a SIGFPE crash, if endpoint_bytes_per_packet is 0 here:

```c
/* Transfers will be at most one frame long: Divide the maximum frame size
         * by the size of the endpoint and round up */
        packets_per_transfer = (ctrl->dwMaxVideoFrameSize +
                                endpoint_bytes_per_packet - 1) / endpoint_bytes_per_packet;
```

I've seen SIGFPE triggered in a real code, but on an invalid ctrl - that's why not proposing a PR right away. But might still be worth changing:

```diff
diff --git a/src/stream.cpp b/src/stream.cpp
index c2c6ff4..bad3013 100644
--- a/src/stream.cpp
+++ b/src/stream.cpp
@@ -1156,7 +1156,7 @@ uvc_error_t uvc_stream_start(
         }
       }
 
-      if (endpoint_bytes_per_packet >= config_bytes_per_packet) {
+      if ((endpoint_bytes_per_packet > 0) && (endpoint_bytes_per_packet >= config_bytes_per_packet)) {
         /* Transfers will be at most one frame long: Divide the maximum frame size
          * by the size of the endpoint and round up */
         packets_per_transfer = (ctrl->dwMaxVideoFrameSize +
```

drinkcat, 2026-11-13:
> Written by Claude (Claude Code, an AI assistant) for the account owner. I found this with [emuvc](https://github.com/drinkcat/emuvc), a UVC camera emulator that runs libuvc against emulated devices through a libusb shim (`LD_PRELOAD`).

Here is a concrete trigger, reproduced on master (f1434af) with an emulated isochronous camera (emuvc config `tests/libuvc/configs/zero-max-payload.toml`).

The device returns `dwMaxPayloadTransferSize = 0` in the probe/commit result. Then in `uvc_stream_start()`:

- `config_bytes_per_packet` is 0.
- The alt-setting loop starts at alt 0, the zero-bandwidth setting with no endpoints, so `endpoint_bytes_per_packet` stays 0.
- `0 >= 0` matches, and `(dwMaxVideoFrameSize + 0 - 1) / 0` at `src/stream.c:1230` raises SIGFPE.

So the camera doesn't need an invalid endpoint descriptor to hit this. A 0 payload size in the negotiated control is enough. Skipping alt settings with `endpoint_bytes_per_packet == 0`, as suggested above, would make libuvc pick the first alt setting that has bandwidth instead of crashing.

To reproduce, from an emuvc checkout:

```sh
scripts/libuvc-test.sh -v -k zero-max-payload
```

drinkcat, 2026-11-14:
(long time coming, thanks for the report...)

closed 2026-11-14 by drinkcat