libuvc/libuvc #194: libuvc on mac can't request camera access

seaCheng, 2021-09-16:
when i use qcamera open my webcam camera, app will pop up a box request camera access due to NSCameraUsageDescription in plist,but when i use libuvc it never do this ? i don't understand why? and i don't konw what to do to make it pop up the same box

mcuee, 2021-09-29:
Is this macOS 12 or macOS 11?

Ref: https://github.com/libuvc/libuvc/issues/188

mcuee, 2021-09-29:
I tend to believe you may have to develop your application using the native macOS API and not libusb/libuvc in this case, if you want to achieve what you want.

libusb under macOS 12 may require another 
Ref: https://github.com/libusb/libusb/pull/911
`kUSBReEnumerateCaptureDeviceMask` requires either the application to be running as root OR the application has the entitlement `com.apple.vm.device-access` AND the user has accepted the authorization for the capture with `IOServiceAuthorize`.

mcuee, 2021-10-30:
You need to use libusb git head and also be an apple developer in order to use the new feature for libusb. You need to develop a GUI app to use the entitlement. For command line version, you may have to use root.

Discussion here.
https://github.com/libusb/libusb/issues/1014#issuecomment-955143331

FAQ here:
https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device-and-claim-exclusive-access

Updates on Oct 2021: with the merging of libusb pull request #911, support for detaching kernel drivers with authorization has been added. This will be included in the upcoming 1.0.25 release. Take note the underlying macOS capture APIs only work on the whole device and not on individual interfaces. So this will force all the kernel extensions (drivers) bound to all the interfaces of a USB Composite devices to be released (Issue #920). You will need to get the entitlement from Apple Developer support, create a provisioning profile with that entitlement, and build your app with that profile. Please take note that command line apps cannot use provisioning profiles and therefore cannot hold this entitlement.

mcuee, 2021-10-30:
More discussions here.
https://github.com/libusb/libusb/issues/1014

mcuee, 2021-11-04:
Based on feedbacks from the libusb discussions, libuvc is not the right library to use under macOS.
https://github.com/libuvc/libuvc/issues/194

mcuee, 2021-11-05:
Please refer to https://github.com/libuvc/libuvc/issues/188

drinkcat, 2026-11-14:
Yeah, MacOS is... complicated. Basically you have to run as root, or request the Apple entitlements (which can only be provided if you are the device manufacturer AFAIU).

closed 2026-11-14 by drinkcat