libuvc/libuvc #196: Camera capture in a single thread

dmikushin, 2021-10-01:
Is it possible to organize USB camera frame grabbing in the main thread?

This design should be a good practice for simple embedded devices with limited multithreading. Also, suppose that you do Python bindings, which do not allow using data directly from a different thread.

yamori813, 2022-02-28:
I seem libuvc is need os. I think you can refer this page.

http://bytewerx.blogspot.com/2009/06/interfacing-usb-webcam-to-pic32.html

dmikushin, 2022-03-01:
Hi @yamori813 btw, do you have experience with pic32? Could you please help me to identify if some particular firmware and chip are indeed pic32>

yamori813, 2022-03-01:
I use PIC32 some times. Now I use PlatformIO make PIC32 firmware.

drinkcat, 2026-11-14:
I think that's out of scope for libuvc, at least not with the current project state (I'm mostly trying to fix bugs, not do complex refactor or add new features).

closed 2026-11-14 by drinkcat