libuvc/libuvc #203: libusb_submit_transfer is caused the phone restart

zklyzhoutao, 2021-11-04:
Hi, We use libuvc and libusb in the Android devices.
When We test the Android device of Google Pixel 4, The App is caused phone restart.
End We find libusb_submit_transfer is get stuck and cause phone restart.

libusb version https://github.com/xloem/libusb/tree/android-dev.
libuvc version 0.06;
Android device: Google Pixel 4 and Android 11.

jksiezni, 2021-11-05:
Hi, it appears to be a common problem on Pixel phones. Users of those devices report identical issues: https://www.reddit.com/r/Pixel4a/comments/oq62kd/usb_otg_webcam_forces_phone_reboot/

You might have just found what function hits a bug. Yet, if it causes restarts, then it's a kernel issue not the library.

zklyzhoutao, 2021-11-06:
Thanks.I got it!
I also find an other question. In phones (like HUAWEI Nova2s ,XiaoMi 9, Galaxy S20),When I first use the libusb library, the code `int r = ioctl(hpriv->fd, IOCTL_USBFS_SUBMITURB, urbs[i])` return -1. When I restart phone, it work good. What settings am I missing？How can I improve this question. 
Thanks!

mcuee, 2026-09-06:
Yes, if the phone restarts, then it is not a libusb/libuvc issue.

closed 2026-09-06 by mcuee