libuvc/libuvc #204: get bytes error in android

zhouconglong, 2021-11-08:
Hi, I use this library in my app.
When I use it, I find I don't get bytes. Well, I restart phone, It is Ok.
And I debugg, I find the  libusb_submit_transfer function return -1.
I think the bytes is big, I edit the endpoint_bytes_per_packet to 1024, it is ok, but only get a part bytes, not a fully image. 
If I want to  get the fully image. I must restart phone.  
Can I lack some code or option?

mcuee, 2022-05-27:
Not so sure if this is a Android kernel problem or libusb or libuvc problem.

Ref: similar reports of libusb Android isoc issues
https://github.com/libusb/libusb/issues?q=is%3Aopen+label%3Aisoc+label%3Aandroid

drinkcat, 2026-11-14:
Old bug, we'd need a lot more details to figure out what's going on. Feel free to open a new bug.

closed 2026-11-14 by drinkcat