libuvc/libuvc #213: how to handle hot plug issue,pull out the camera USB when the camera is running

shaozhihua, 2022-03-16:
sometimes we may pull out the camera USB on occasionaly,we dont do that on purpose,what confused me is we plug/pull out the USB if the camera is running well,so how should we monitor this issue and solve it,I attempted judge it in callback function ,but when we pull out the USB,the program wont call the cb......
I need help..sir or ladies...

shaozhihua, 2022-03-16:
sometimes we may pull out the camera USB on occasionaly,we dont do that on purpose,what confused me is we plug/pull out the USB if the camera is running well,so how should we monitor this issue and solve it,I attempted judge it in callback function ,but when we pull out the USB,the program wont call the cb......

mcuee, 2022-05-08:
libusb does support hotplug under macOS and Linux (but not on Windows), probably libuvc can be enhanced to support hotplug.

shaozhihua, 2022-05-23:
Ok then,Isee,thanks for the telling,I want to find another way to solve it,like monitor the driver and give a warn when pull out the camera USB driver;

shaozhihua, 2022-06-22:
after all does not support on windows,I accept the result;

closed 2022-06-22 by shaozhihua