libuvc/libuvc #216: uvc_init return -1 in android 10

751643992, 2022-04-26:
compile https://github.com/libusb/libusb
compile https://github.com/libjpeg-turbo/libjpeg-turbo
uvc_init return -1 in android 10? 
Can libuvc run in android platform?
But https://github.com/saki4510t/UVCCamera,  this project have modied libusb and libuvc，uvc_init2。

mcuee, 2022-05-08:
You need to try the following patch to libusb if your device is not rooted.
https://github.com/libusb/libusb/pull/874

Caprey, 2023-05-16:
Did you solve this problem, I also encountered the same problem, in the android-example of libuvc uvc_init return -1
```c++
JNIEXPORT void JNICALL
Java_org_example_ExampleJava_nativeOpenUVC(JNIEnv *env, jobject thiz, jint sys_dev)
{
	uvc_context_t *uvcContext = nullptr;

	uvc_error_t res = uvc_init(&uvcContext, nullptr);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_init() failed: %s", uvc_strerror(res));
		return;
	}

	uvc_device_handle_t *dev = nullptr;
	res = uvc_wrap(sys_dev, uvcContext, &dev);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_wrap() failed: %s", uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}

	uvc_stream_ctrl_t ctrl;
	res = uvc_get_stream_ctrl_format_size(dev, &ctrl, UVC_FRAME_FORMAT_ANY, 640, 480, 30);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_get_stream_ctrl_format_size(%dx%d@%d) failed: %s", _width, _height, _fps, uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}

	uvc_stream_handle_t *stream = nullptr;
	res = uvc_stream_open_ctrl(dev, &stream, &ctrl);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_stream_open_ctrl() failed: %s", uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}

	res = uvc_stream_start(stream, captureCallback, nullptr, 0);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_stream_start() failed: %s", uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}
}
```

Caprey, 2023-05-16:
I have solved it. It is necessary to disable the USB detection before uvc_init() by calling the libusb_set_option(&ctx, LIBUSB_OPTION_WEAK_AUTHORITY, NULL) method.
Related link: https://github.com/libusb/libusb/blob/master/android/examples/unrooted_android.c

```c++
JNIEXPORT void JNICALL
Java_org_example_ExampleJava_nativeOpenUVC(JNIEnv *env, jobject thiz, jint sys_dev)
{
        int ret = libusb_set_option(&ctx, LIBUSB_OPTION_WEAK_AUTHORITY, NULL);
        if (ret != LIBUSB_SUCCESS) {
             LOG_ERROR("libusb_set_option failed: %d\n", ret);
             return;
        }

	uvc_context_t *uvcContext = nullptr;

	uvc_error_t res = uvc_init(&uvcContext, nullptr);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_init() failed: %s", uvc_strerror(res));
		return;
	}

	uvc_device_handle_t *dev = nullptr;
	res = uvc_wrap(sys_dev, uvcContext, &dev);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_wrap() failed: %s", uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}

	uvc_stream_ctrl_t ctrl;
	res = uvc_get_stream_ctrl_format_size(dev, &ctrl, UVC_FRAME_FORMAT_ANY, 640, 480, 30);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_get_stream_ctrl_format_size(%dx%d@%d) failed: %s", _width, _height, _fps, uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}

	uvc_stream_handle_t *stream = nullptr;
	res = uvc_stream_open_ctrl(dev, &stream, &ctrl);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_stream_open_ctrl() failed: %s", uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}

	res = uvc_stream_start(stream, captureCallback, nullptr, 0);
	if(res != UVC_SUCCESS) {
		LOG_ERROR("uvc_stream_start() failed: %s", uvc_strerror(res));
		uvc_exit(uvcContext);
		return;
	}
}
```

Junzz0, 2026-09-13:
Runs fine on Android.

It's recommended to run the `unrooted_usb_description` function on Android.
https://github.com/libusb/libusb/blob/master/android/examples/unrooted_android.c

Can't forget to set it **LIBUSB_OPTION_NO_DEVICE_DISCOVERY**

drinkcat, 2026-11-14:
Yep. I just added an example that should hopefully help people who want to use libuvc on Android.

closed 2026-11-14 by drinkcat