libuvc/libuvc #227: After reopening the same uvc device streaming does not work any more.

damijans, 2022-07-28:
The problem can be easily reproduced by modifying example.c and embedding part of the code from uvc_open() to uvc_close() into while loop:

int main()
{ 
    ....
    res = uvc_init(&ctx, NULL);
    ....
    while (true) {
         ...
         res = uvc_open(dev, &devh);
         ... 
         res = uvc_start_streaming(devh, &ctrl, cb, (void *) 12345, 0);
         ....
         uvc_stop_streaming(devh);
         ...
         uvc_close(devh);
    }
     uvc_exit(ctx);
}

In the first loop iteration streaming works correctly, while in the second iteration, streaming is not started and later on in uvc_stop_streamin() function the application hangs.

Note there exist several tickets already complaining about streaming does not work or the application hangs in uvc_stop_streaming() function, for example 63, 53, 16, 152...

I always reproduce this issue on Ubuntu 16.04 with the code from master branch with commit d3318ae72a2b916ae352ad0abbbfa2c0990f455e.

damijans, 2022-07-28:
Streaming in libuvc library is implemented by using USB ASYNCHRONOUS I/O with the help of libusb_submit_transfer() function. As a result of calling this function after transfer is finished _uvc_stream_callback() function is invoked (pointer to this function is specified as an argument while calling libusb_transfer() function).  Within _uvc_stream_callback() function libusb_submit_transfer() function is called again, what somehow implements a loop of asynchronous transfers without having some special thread to execute a loop.

However in order to assure this mechanism works there must be someone who periodically calls libusb_handle_events_completed() ( or some alternative function for handling pending events). In libuvc this is done in separate thread (_uvc_handle_events), which is invoked when the first device is opened and terminated when the last opened device is closed. Terminating this thread is implemented with the help of ctx->kill_handler_thread which  is set from uvc_close() function. The problem which cause complained behavior is that this flag remains set also next time when _uvc_handle_events() thread is invoked again, what cause that thread it is immediately stopped. As a result libusb_handle_events_completed() is never called and consequently USB ASYNCHRONOUS I/O  streaming does not work any more.

The problem was solved by clearing  ctx->kill_handler_thread flag just before _uvc_handle_events() thread is started.

FrancoisDn, 2023-09-12:
Thank you for your fix!
I face the same problem, and with your commit, it fixes the problem on Linux and also on Android!