libuvc/libuvc #229: UVC_FRAME_FORMAT_GRAY8 GUID

RudyB, 2022-07-29:
I am using a UVC compatible camera that provides 8 bit and 12 bit grayscale images. Initially I could not connect to it via libuvc because the guid that the camera provides is slightly different from the libuvc guid `UVC_FRAME_FORMAT_GRAY8` . The only difference is that my camera has spaces instead of 0's after the Y8. I was eventually able to get libuvc to work by changing the source code but I was wondering if there was a better way of connecting to cameras with a frame format that is not already in the libuvc codebase. 

libuvc UVC_FRAME_FORMAT_GRAY8 GUID
`    FMT(UVC_FRAME_FORMAT_GRAY8,
      {'Y',  '8',  '0',  '0', 0x00, 0x00, 0x10, 0x00, 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71})`

camera UVC_FRAME_FORMAT_GRAY8 GUID
`    FMT(UVC_FRAME_FORMAT_GRAY8,
      {'Y',  '8',  ' ',  ' ', 0x00, 0x00, 0x10, 0x00, 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71})`

FrostKiwi, 2022-12-06:
> to cameras with a frame format that is not already in the libuvc codebase.

Same question from me. Great that you figured it out for you use case.
Same with NV21 vs NV12 as mentioned in https://github.com/libuvc/libuvc/issues/240 I imagine.

I also couldn't use a fork of libUVC, because it didn't have NV12 support and I required it, so had to build the master libUVC myself, as mentioned in https://github.com/libuvc/libuvc/issues/245

drinkcat, 2026-11-14:
Took a while ,-P Thanks for the report, fixed!

closed 2026-11-14 by drinkcat