libuvc/libuvc #233: Error building the test program under macOS

mcuee, 2022-09-05:
Sorry I do not know much about CMake and I am not so sure how to fix the issue.

```
mcuee@mcuees-Mac-mini libuvc % mkdir build_darwin
mcuee@mcuees-Mac-mini libuvc % cd build_darwin 
mcuee@mcuees-Mac-mini build_darwin % cmake .. -DBUILD_TEST=ON -DBUILD_EXAMPLE=ON  
-- The C compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- No build type selected, default to Release
-- No target type selected, default to both shared and static library
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.26
-- libusb-1.0 found using pkgconfig
-- Found JPEG: /opt/homebrew/lib/libjpeg.dylib (found version "80") 
-- Found JPEG library using standard module
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.24.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (JPEG) does
  not match the name of the calling package (JpegPkg).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindJpegPkg.cmake:58 (find_package_handle_standard_args)
  CMakeLists.txt:45 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found JPEG: /opt/homebrew/lib/libjpeg.dylib  
-- Building libuvc with JPEG support.
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Components requested: opencv_core;opencv_highgui
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/libusb/libuvc/libuvc/build_darwin

mcuee@mcuees-Mac-mini build_darwin % make
[  4%] Building C object CMakeFiles/uvc.dir/src/ctrl.c.o
[  8%] Building C object CMakeFiles/uvc.dir/src/ctrl-gen.c.o
[ 12%] Building C object CMakeFiles/uvc.dir/src/device.c.o
[ 16%] Building C object CMakeFiles/uvc.dir/src/diag.c.o
[ 20%] Building C object CMakeFiles/uvc.dir/src/frame.c.o
[ 25%] Building C object CMakeFiles/uvc.dir/src/init.c.o
[ 29%] Building C object CMakeFiles/uvc.dir/src/stream.c.o
[ 33%] Building C object CMakeFiles/uvc.dir/src/misc.c.o
[ 37%] Building C object CMakeFiles/uvc.dir/src/frame-mjpeg.c.o
[ 41%] Linking C shared library libuvc.dylib
[ 41%] Built target uvc
[ 45%] Building C object CMakeFiles/uvc_static.dir/src/ctrl.c.o
[ 50%] Building C object CMakeFiles/uvc_static.dir/src/ctrl-gen.c.o
[ 54%] Building C object CMakeFiles/uvc_static.dir/src/device.c.o
[ 58%] Building C object CMakeFiles/uvc_static.dir/src/diag.c.o
[ 62%] Building C object CMakeFiles/uvc_static.dir/src/frame.c.o
[ 66%] Building C object CMakeFiles/uvc_static.dir/src/init.c.o
[ 70%] Building C object CMakeFiles/uvc_static.dir/src/stream.c.o
[ 75%] Building C object CMakeFiles/uvc_static.dir/src/misc.c.o
[ 79%] Building C object CMakeFiles/uvc_static.dir/src/frame-mjpeg.c.o
[ 83%] Linking C static library libuvc.a
[ 83%] Built target uvc_static
[ 87%] Building C object CMakeFiles/example.dir/src/example.c.o
/Users/mcuee/build/libusb/libuvc/libuvc/src/example.c:26:74: warning: cast to smaller integer type 'int' from 'void *' [-Wvoid-pointer-to-int-cast]
    frame->frame_format, frame->width, frame->height, frame->data_bytes, (int) ptr);
                                                                         ^~~~~~~~~
1 warning generated.
[ 91%] Linking C executable example
[ 91%] Built target example
[ 95%] Building C object CMakeFiles/uvc_test.dir/src/test.c.o
In file included from /Users/mcuee/build/libusb/libuvc/libuvc/src/test.c:35:
In file included from /opt/homebrew/Cellar/opencv/4.6.0/include/opencv4/opencv2/highgui/highgui_c.h:45:
In file included from /opt/homebrew/Cellar/opencv/4.6.0/include/opencv4/opencv2/core/core_c.h:48:
In file included from /opt/homebrew/Cellar/opencv/4.6.0/include/opencv4/opencv2/core/types_c.h:84:
/opt/homebrew/Cellar/opencv/4.6.0/include/opencv4/opencv2/core/cvdef.h:775:4: error: "OpenCV 4.x+ requires enabled C++11 support"
#  error "OpenCV 4.x+ requires enabled C++11 support"
   ^
/opt/homebrew/Cellar/opencv/4.6.0/include/opencv4/opencv2/core/cvdef.h:781:10: fatal error: 'array' file not found
#include <array>
         ^~~~~~~
2 errors generated.
make[2]: *** [CMakeFiles/uvc_test.dir/src/test.c.o] Error 1
make[1]: *** [CMakeFiles/uvc_test.dir/all] Error 2
make: *** [all] Error 2

```

mcuee, 2022-09-05:
Tried the tip here but it does no seem to work.
Ref: https://stackoverflow.com/questions/10851247/how-do-i-activate-c-11-in-cmake

```

mcuee@mcuees-Mac-mini libuvc % git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 87ae8f7..ae2a835 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -4,6 +4,13 @@ project(libuvc
   LANGUAGES C
 )
 
+set (CMAKE_CXX_STANDARD 11)
+
+# Fix behavior of CMAKE_CXX_STANDARD when targeting macOS.
+if (POLICY CMP0025)
+  cmake_policy(SET CMP0025 NEW)
+endif ()
+
 # Additional search scripts path for libusb-1.0, libjpeg, OpenCV
 list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_LIST_DIR}/cmake)

```

papr, 2022-09-14:
@mcuee I think the issue is that this step attempts to build a C object, not a c++ object.

If you run `cmake --build . --clean-first -v`, then you will see that `cc` is being called, not `c++`.

papr, 2022-09-14:
From what I understand, OpenCV's C API is no longer supported. One would probably need to port the test to a dedicated cxx-cmake project and the corresponding c++ OpenCV api.

mcuee, 2023-05-16:
Same problem here.
* https://github.com/libuvc/libuvc/issues/254

drinkcat, 2026-09-05:
https://github.com/libuvc/libuvc/pull/307 will fix this.

closed 2026-09-06 by drinkcat

mcuee, 2026-09-06:
Great.

Indeed we can see that #307 will fix this issue from github action run.
https://github.com/libuvc/libuvc/actions/runs/34004994905/job/101410442306?pr=307