libuvc/libuvc #239: publish uvc_frame_format_for_guid() function

damijans, 2022-09-28:
uvc_frame_format_for_guid() function is used to retrieve frame type based on specified frame GUID identifier. This function is currently used internally within libuvc but it would be wise to make it public. Namely, activating certain frame type is based on specifying frame type while calling uvc_get_stream_ctrl_format_size() function. This frame type must be "known" to the caller of a function, however libuvc does not 
publish this function.

damijans, 2022-09-28:
Suggested enhancement is implemented and is waiting as one of changes in pull requests https://github.com/libuvc/libuvc/pull/224

drinkcat, 2026-09-06:
Also pullded in https://github.com/libuvc/libuvc/pull/314

closed 2026-09-06 by drinkcat