libuvc/libuvc #240: Make libuvc to be aware of NV21 & I420 GUID

damijans, 2022-09-28:
Currently libuvc is not aware of two guid values, which are available on one of cameras I was working with. Please, integrate the following guid

{'I',  '4',  '2',  '0', 0x00, 0x00, 0x10, 0x00, 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}
{'N',  'V',  '2',  '1', 0x00, 0x00, 0x10, 0x00, 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}

damijans, 2022-09-28:
proposed functionality is prepared and is waiting for review in pull request https://github.com/libuvc/libuvc/pull/224

drinkcat, 2026-09-05:
Oh, I missed that change in the PR, I'll separate it out.

Those 2:
https://github.com/libuvc/libuvc/pull/224/changes/a8d6eeb9efdd5a75d69b877cbf213985f2bca909
https://github.com/libuvc/libuvc/pull/224/changes/a89187448700b58c88884e964ffd8381af434120

drinkcat, 2026-09-06:
Pulled in https://github.com/libuvc/libuvc/pull/314.

closed 2026-09-06 by drinkcat