libuvc/libuvc #241: How could I open the two child devices by libuvc?

andymatos, 2022-09-30:
I have two uvc video cameras under one usb port. How could I open them at same time?
I try to get two different uvc_stream_ctrls , and call uvc_start_streaming with these two different ctrls. First is open correctly, but for second, is failed with Busy (-6) error. Since they're using the same interface.
This device works fine on Winwos with direct show api. Just do not know how to make it work in libuvc.
The lsusb -t show as follow:
[image]
For v4l2, it create the /dev/vide0 ~ /dev/video4, and /dev/video0 and /dev/video2 could open at same time by vlc

jksiezni, 2022-10-04:
The libuvc was not designed to support this scenario, so you would need to alter streaming code to start multiple streams. However, you could make a workaround by creating two uvc_contexts and open the device twice.
```
uvc_context_t *ctx1, *ctx2;
uvc_init(&ctx1, NULL);
uvc_init(&ctx2, NULL);
```
Anyway, this is rather unusual for a device to have multiple Video/Audio interfaces, so could you say what kind of device is it?

andymatos, 2022-10-05:
Hi, jksiezni, thanks for your reply.
After uvc_init with two different context, how to open it twice? As far as what I have tried, the second uvc_open will return with UVC_ERROR_BUSY.
This is a device come from a hardware team which use the chipset from the MStar. It used to be used to send mulit stream for the TV Box, and we use it to send the strem in uvc protocol.

I have looked into the libuvc code, I found the code is failed with _uvc_get_stream_by_interface when call the uvc_start_streaming. If I could map the different interface when start it at second time, will that work?

jksiezni, 2022-10-05:
Oh, I see what your issue might be then. So, forget what I said earlier for a moment.
You said that you start streams from the same interface, so that's why you get BUSY errors around `_uvc_get_stream_by_interface()`. You have to make sure that each stream is configured on a different interface number.
It is configured via `uvc_stream_ctrl_t.bInterfaceNumber` field.

Also note, configuring streams via `uvc_get_stream_ctrl_format_size()` very likely won't work for you, especially if image formats repeat for Video Stream Interfaces. So you will need to implement your own version that iterates through selected stream interface only. Reading a UVC specification might come in handy.

If you don't want to mess up with libuvc and know your hardware very well, then you could setup `uvc_stream_ctrl_t` and probe `uvc_probe_stream_ctrl()` all by yourself.

andymatos, 2022-10-08:
Thanks again for your reply.
I will look into the UVC specification and try to make some changes on the flow in the libuvc to map the second stream on a different interface

closed 2022-10-08 by andymatos