libuvc/libuvc #242: get_mode: Invalid mode (-51)

U-Mallik, 2022-10-22:
When I use res = uvc_get_stream_ctrl_format_size 

My camera throws the following error. "get_mode: Invalid mode (-51)". I am using the stream.c routine from the downloaded floder. I have tried 

frame_format = UVC_FRAME_FORMAT_MJPEG; //UVC_FRAME_FORMAT_GRAY8; //UVC_FRAME_FORMAT_BY8; //UVC_FRAME_FORMAT_YUYV;

to no avail. I am using an FT602 device to read in image data. I see that the UVC device is identifying itself correctly - (0403:602a/1)
I see that the format (GREY), the frame size 1280x1024 and the Frame Rate (30) are correct. Any suggestions?

UVC initialized
Device found
Device opened
DEVICE CONFIGURATION (0403:602a/1) ---
Status: idle
VideoControl:
	bcdUVC: 0x0110
VideoStreaming(1):
	bEndpointAddress: 129
	Formats:
	UncompressedFormat(1)
		  bits per pixel: 8
		  GUID: 4752455900001000800000aa00389b71 (GREY)
		  default frame: 1
		  aspect ratio: 0x0
		  interlace flags: 00
		  copy protect: 00
			FrameDescriptor(1)
			  capabilities: 00
			  size: 1280x1024
			  bit rate: 16-400000000
			  max frame size: 1310720
			  default interval: 1/30
			  interval[0]: 1/30
END DEVICE CONFIGURATION

First format: (GREY) 1280x1024 30fps
bmHint: 0000
bFormatIndex: 0
bFrameIndex: 0
dwFrameInterval: 0
wKeyFrameRate: 23504
wPFrameRate: 37651
wCompQuality: 21990
wCompWindowSize: 0
wDelay: 42064
dwMaxVideoFrameSize: 32765
dwMaxPayloadTransferSize: 3109459296
bInterfaceNumber: 253
get_mode: Invalid mode (-51)
Device closed
UVC exited

damijans, 2022-10-31:
Hi,

I don't know details about your camera, however I had (still have) the same issue with my camera. By investigating the source code of uvc_get_stream_ctrl_format_size() function I realized that it does not follow UVC 1.5 class specification  where in "4.3.1.1 Video Probe and Commit Controls" section it is depicted that selection of video format is based on negotiation between host and video streaming interfaces. Practically that would mean that negotiation compare capabilities of USB port on host system and Camera requirements. In libuvc library I have found negotiation is comparing actual requirements of target video format towards maximal requirements for currently selected frame format and this is completely something different.

For a couple of camera I was working on I have found a workaround to call uvc_get_stream_ctrl_format_size() function twice and ignoring the first result. Unfortunately this does not work for all cameras I have.

drinkcat, 2026-11-15:
Aware it's an old bug, but what would be most useful here is some usbmon traces, so that we can understand more exactly how the device (mis?)behaves.