libuvc/libuvc #245: Windows + MinGW64 in November 2022

FrostKiwi, 2022-11-26:
Used libUVC with great success out of the box on Linux, moved to Windows and it was a painful 6 hours long debug session. Want to quickly sumarize how it ended up and what the result looks like. Since this is not an issue, I'll just close it in a week, just want to document how I got things running in a MinGW64 environment.

I just need one demonstration with 2 cameras hooked up to two Elgato CamLinks up and running within 24h, so I ended up doing a lot of hacky things.

@mcuee Has been doing god's work, linking to vital resources https://github.com/pupil-labs/libuvc/issues/10#issuecomment-403290068 and distributing prepuilt libraries https://github.com/libuvc/libuvc/issues/12#issuecomment-1120397265. Directly linking his MSVC built dlls from MinGW64's GCC results in segfaults on every function that prints to the console. Ignoring those functions, there is a deal breaker for me: https://github.com/pupil-labs/libuvc did not support NV12 as I found out, the only mode, that support 4k streaming on the Cam Link. (Also a super nice to work with format, match made in heaven for Shader access). Same goes for https://github.com/rageworx/libuvc. It's a repo made for MinGW64, but grepping through it did not match for NV12, therefore I just skiped trying that one. So I had to make the upstream https://github.com/libuvc/libuvc work. (There is also I420, but neither repo supports this one, as reported recently by @damijans in https://github.com/libuvc/libuvc/issues/240)

A big help was this french blog I came across, who built libUVC on MinGW64: https://anthony.lepors.fr/raspi-thermo-cam/compilation-libusb-libuvc-msys2/ (didn't even need deepl.com , really well put together blog with images and file previews ), though I ended up doing this a bit differently.

This repo has misconfigured CMake build instructions if used in combination with MSYS2 for linking libUSB. There have been multiple solutions proposed and multiple repos have provided patches, but nothing made it to upstream. 
I just commented out the libUSB part in CMakeLists.txt.
```
  # target_link_libraries(${target_name}
    # libusb-1.0 used internally so we link to it privately.
   # PRIVATE LibUSB::LibUSB
  #)
```
If you use MinGW64, then you don't need to compile libUSB, because MSYS2's package manager has it already in store: `pacman -S mingw-w64-x86_64-libusb`
I generated Ninja build files, hand copied both libUSB header and library `cp /mingw64/lib/libusb-1.0.dll.a libuvc/build/CMakeFiles/uvc.dir/src` `cp /mingw64/include/libusb-1.0/libusb.h libuvc/build/include/libusb.h` and edited the link step of the auto generated Ninja build.ninja file to append the lib file:
```
#############################################
# Link the shared library libuvc.dll

build libuvc.dll libuvc.dll.a: C_SHARED_LIBRARY_LINKER__uvc_Release CMakeFiles/uvc.dir/src/ctrl.c.obj CMakeFiles/uvc.dir/src/ctrl-gen.c.obj CMakeFiles/uvc.dir/src/device.c.obj CMakeFiles/uvc.dir/src/diag.c.obj CMakeFiles/uvc.dir/src/frame.c.obj CMakeFiles/uvc.dir/src/init.c.obj CMakeFiles/uvc.dir/src/stream.c.obj CMakeFiles/uvc.dir/src/misc.c.obj CMakeFiles/uvc.dir/src/libusb-1.0.dll.a
  LANGUAGE_COMPILE_FLAGS = -O3 -DNDEBUG
[...]
```

Linking just the resulting libuvc.dll.a to my final program was not enough, because the build file was configured for Shared build mode. It had to be both libuvc.dll.a and libuvc.dll. Normally a deal breaker, but it's just a one of demo, I'm sure it can be reconfigured for proper static only linkage.

Next, the default Camera driver won't work. Following https://github.com/pupil-labs/pyuvc/blob/master/WINDOWS_USER.md steps 1-6, I replaced both the CamLink's drivers with libusbK. This makes libUVC play along, but as a side effect, OBS does not recognize those CamLink's anymore. What a pain, have to switch back the drivers. (Which makes me wonder, whether each driver switch fills up the C: drive with System Restore points...)

The result:
<details><summary>Device Config of a CamLink with a Panasonic GH6 in 4K mode</summary>

```
DEVICE CONFIGURATION (0fd9:0066/0005AC85A6000) ---
Status: idle
VideoControl:
        bcdUVC: 0x0110
VideoStreaming(1):
        bEndpointAddress: 131
        Formats:
        UncompressedFormat(1)
                  bits per pixel: 12
                  GUID: 4e56313200001000800000aa00389b71 (NV12)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 3840x2160
                          bit rate: -1308983296--1308983296
                          max frame size: 12441600
                          default interval: 1/29
                          interval[0]: 1/29
        UncompressedFormat(2)
                  bits per pixel: 12
                  GUID: 4e56313200001000800000aa00389b71 (NV12)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 3840x2160
                          bit rate: -1308983296--1308983296
                          max frame size: 12441600
                          default interval: 1/29
                          interval[0]: 1/29
[... 53 lines cut ...]
                          default interval: 1/50
                          interval[0]: 1/50
        UncompressedFormat(3)
                  bits per pixel: 12
                  GUID: 4934323000001000800000aa00389b71 (I420)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 1920x1080
                          bit rate: 1244160000-1244160000
                          max frame size: 3110400
                          default interval: 1/50
                          interval[0]: 1/50
END DEVICE CONFIGURATION
```
</details>
Both cameras work, but their behaviour is strange, as compared to Linux.
The 4k camera in NV12 mode should stream at 30fps, but libUVC only updates at 24fps, regardless of circumstances. (I confirmed the camera streaming at 30 fps in OBS before switching drivers)

Then the second 1080p Camera does not enter NV12 mode, but fails with `get_mode: Invalid mode (-51)`. This one I don't understand at all. It's fine in Linux. It's fine in YUV2 mode though on Windows. Even though it has way less data to stream, it only streams at 4.8fps. Again the strange FPS shift. So I ended up with the YUV2 mode, uvc_yuyv2rgb() conversion and a 4.8 fps framerate on the 2nd Cam. Mind you, the results are the same whether or not both cameras stream at the same time or not.

The main camera streams at 24 fps and the secondary overlayed one is not that important for my demo, I'm happy with what I got. But man is all of this unrelaible as compared to Linux. Across two Windows machines, same result. Propably libusbK acting up.

FrostKiwi, 2022-11-26:
Ohh... And just in case any of you were wonder on how reliable the drivers are...

https://user-images.githubusercontent.com/60887273/204079068-a2406414-691d-4c2c-9aaf-b286db76fe71.mp4

(Finger pointed at Loose HDMI connector)

FrostKiwi, 2022-12-06:
> Again the strange FPS shift. So I ended up with the YUV2 mode, uvc_yuyv2rgb() conversion and a 4.8 fps framerate on the 2nd Cam. 

I don't know why, but I replaced the HDMI cable for my on stage demo and my streamed FPS shot up to the full 50fps on the second cam, that was 5 fps before. O.O
Maybe a coincidence. Either way everything worked perfectly in the end.
That was my experience with libUVC on Windows.

Sry for the long blog like mess, thought I bundle all this information, maybe useful to someone.

closed 2022-12-06 by FrostKiwi

mcuee, 2022-12-30:
@FrostKiwi 
Thanks a lot for the detailed instructions. I am able to follow your instructions and build libuvc git HEAD under MSYS2/mingw64.

Result binary seems to work fine.
https://github.com/libuvc/libuvc/issues/12#issuecomment-1367972518