libuvc/libuvc #252: uvc_test shared library not generating

asadasivam, 2023-02-21:
I am following the commands from [https://libuvc.github.io/libuvc/index.html](url)
`$ git clone https://github.com/libuvc/libuvc.git
$ cd libuvc
$ mkdir build
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release ..
$ make && make install`

Only example.so file generating. uvc_test code not generated...

asadasivam, 2023-02-21:
> error: -- libusb-1.0 found using pkgconfig
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so (found version "80") 
-- Found JPEG library using standard module
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Building libuvc with JPEG support.
-- Components requested: opencv_core;opencv_highgui
-- Configuring done
CMake Error in CMakeLists.txt:
  Cannot use features from non-enabled language CXX

CMake Generate step failed.  Build files cannot be regenerated correctly.

mcuee, 2023-03-05:
It seems CMake scripts may need some update to enable CXX, since new version of OpenCV requires C++ 11 support.

Build log under Ubuntu 20.04.

```
mcuee@HeroBox:~/build/libusb/libuvc/build$ cmake -DBUILD_TEST=ON -DBUILD_EXAMPLE=ON ..
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- No build type selected, default to Release
-- No target type selected, default to both shared and static library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.26
-- libusb-1.0 found using pkgconfig
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so (found version "80")
-- Found JPEG library using standard module
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so
-- Building libuvc with JPEG support.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Components requested: opencv_core;opencv_highgui
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/build/libusb/libuvc/build
mcuee@HeroBox:~/build/libusb/libuvc/build$ ls
CMakeCache.txt  CMakeFiles  cmake_install.cmake  include  libuvcConfigVersion.cmake  libuvc.pc  Makefile
mcuee@HeroBox:~/build/libusb/libuvc/build$ cmake --build .
Scanning dependencies of target uvc
[  4%] Building C object CMakeFiles/uvc.dir/src/ctrl.c.o
[  8%] Building C object CMakeFiles/uvc.dir/src/ctrl-gen.c.o
[ 12%] Building C object CMakeFiles/uvc.dir/src/device.c.o
[ 16%] Building C object CMakeFiles/uvc.dir/src/diag.c.o
[ 20%] Building C object CMakeFiles/uvc.dir/src/frame.c.o
[ 25%] Building C object CMakeFiles/uvc.dir/src/init.c.o
[ 29%] Building C object CMakeFiles/uvc.dir/src/stream.c.o
[ 33%] Building C object CMakeFiles/uvc.dir/src/misc.c.o
[ 37%] Building C object CMakeFiles/uvc.dir/src/frame-mjpeg.c.o
[ 41%] Linking C shared library libuvc.so
[ 41%] Built target uvc
Scanning dependencies of target uvc_test
[ 45%] Building C object CMakeFiles/uvc_test.dir/src/test.c.o
In file included from /usr/include/opencv4/opencv2/core/types_c.h:84,
                 from /usr/include/opencv4/opencv2/core/core_c.h:48,
                 from /usr/include/opencv4/opencv2/highgui/highgui_c.h:45,
                 from /home/mcuee/build/libusb/libuvc/src/test.c:35:
/usr/include/opencv4/opencv2/core/cvdef.h:704:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"
  704 | #  error "OpenCV 4.x+ requires enabled C++11 support"
      |    ^~~~~
/usr/include/opencv4/opencv2/core/cvdef.h:710:10: fatal error: array: No such file or directory
  710 | #include <array>
      |          ^~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/uvc_test.dir/build.make:63: CMakeFiles/uvc_test.dir/src/test.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:82: CMakeFiles/uvc_test.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

mcuee, 2023-03-05:
Similar issues here.
* https://github.com/Tossy0423/yolov4-for-darknet_ros/issues/13

mcuee, 2023-03-05:
I think this is also the same issue as what I encountered under macOS.
* https://github.com/libuvc/libuvc/issues/233

mcuee, 2023-03-05:
From the comment from @papr. So for now we have to forget the test program.

* https://github.com/libuvc/libuvc/issues/233#issuecomment-1246760664

> From what I understand, OpenCV's C API is no longer supported. One would probably need to port the test to a dedicated cxx-cmake project and the corresponding c++ OpenCV api.

drinkcat, 2026-09-05:
Yeah looks the same as #233

closed 2026-09-05 by drinkcat