libuvc/libuvc #254: Tests fail to compile: error: "OpenCV 4.x+ requires enabled C++11 support"

yurivict, 2023-04-02:
```
-- Build files have been written to: /usr/ports/multimedia/libuvc/work/.build
[ 50% 1/2] /usr/local/libexec/ccache/cc  -I/usr/ports/multimedia/libuvc/work/libuvc-0.0.7/include -I/usr/ports/multimedia/libuvc/work/.build/include -isystem /usr/local/include/opencv4 -O2 -pipe  -fstack-protector-strong -fno-strict-aliasing -O2 -pipe  -fstack-protector-strong -fno-strict-aliasing  -DNDEBUG -MD -MT CMakeFiles/uvc_test.dir/src/test.c.o -MF CMakeFiles/uvc_test.dir/src/test.c.o.d -o CMakeFiles/uvc_test.dir/src/test.c.o -c /usr/ports/multimedia/libuvc/work/libuvc-0.0.7/src/test.c
FAILED: CMakeFiles/uvc_test.dir/src/test.c.o 
/usr/local/libexec/ccache/cc  -I/usr/ports/multimedia/libuvc/work/libuvc-0.0.7/include -I/usr/ports/multimedia/libuvc/work/.build/include -isystem /usr/local/include/opencv4 -O2 -pipe  -fstack-protector-strong -fno-strict-aliasing -O2 -pipe  -fstack-protector-strong -fno-strict-aliasing  -DNDEBUG -MD -MT CMakeFiles/uvc_test.dir/src/test.c.o -MF CMakeFiles/uvc_test.dir/src/test.c.o.d -o CMakeFiles/uvc_test.dir/src/test.c.o -c /usr/ports/multimedia/libuvc/work/libuvc-0.0.7/src/test.c
In file included from /usr/ports/multimedia/libuvc/work/libuvc-0.0.7/src/test.c:35:
In file included from /usr/local/include/opencv4/opencv2/highgui/highgui_c.h:45:
In file included from /usr/local/include/opencv4/opencv2/core/core_c.h:48:
In file included from /usr/local/include/opencv4/opencv2/core/types_c.h:84:
/usr/local/include/opencv4/opencv2/core/cvdef.h:775:4: error: "OpenCV 4.x+ requires enabled C++11 support"
#  error "OpenCV 4.x+ requires enabled C++11 support"
   ^
/usr/local/include/opencv4/opencv2/core/cvdef.h:781:10: fatal error: 'array' file not found
#include <array>
         ^~~~~~~
2 errors generated.
ninja: build stopped: subcommand failed.
*** Error code 1
```

Version: 0.0.7
clang-14
FreeBSD 13.1

jwlodek, 2023-04-04:
I believe you can enable C++11 with

```
set (CMAKE_CXX_STANDARD 11)
```

in the CMakeLists.txt.

drinkcat, 2026-09-05:
#307 will fix this.

drinkcat, 2026-09-05:
Closing, tracking in #233.

closed 2026-09-05 by drinkcat