libuvc/libuvc #257: Using EndPointHalt to Stop UVC Bulk Mode Transfer

yjun123, 2023-05-16:
Hi, I am working on a UVC bulk camera using libuvc in Android. It works fine drived by libuvc . But I found some difference between Android libuvc and Linux V4l2 for UVC bulk mode.

When Stoping UVC Bulk Camera, Linux V4l2 driver will send a std request(ClearFeature) to bulk endpoint(EndpointHalt), but libuvc on Android doesn't.

I just find the reason why Linux send ClearFeature to Bulk mode UVC camera, see below:

1. Linux Commit: [[media] uvcvideo: Do not use usb_set_interface on bulk EP](https://github.com/torvalds/linux/commit/b1e43f232698274871e1358c276d7b0242a7d607)
2. Linux Mail Discussion : https://sourceforge.net/p/linux-uvc/mailman/linux-uvc-devel/thread/5301D32E.400%40rempel-privat.de/#msg31986501
 
Linux just  imitated the reaction of Windows, Windows will send ClearFeature to the Video Streaming Endpoint When Stoping UVC Bulk Camera, too.

So, Could libuvc add new Api to do ClearFeature to VS endpoint instead of Sending SetInterface like UVC ISO Camera?

https://github.com/libuvc/libuvc/blob/68d07a00e11d1944e27b7295ee69673239c00b4b/src/device.c#L1020

if not, how should I do to make reaction of libuvc same as Windows or Linux?

bad English, Sorry!

Thanks!

jksiezni, 2023-05-24:
Yeah, you are right. This is how a bulk stream should be stopped:
https://github.com/nExtCamera/libuvc/blob/dfe81a61ccaac486d1aa5b0d0747324d9ed1896c/src/stream.c#LL1595C1-L1600C4
Although, some bulk devices do not respect CLEAR HALT request, thus I'd keep `set_interface_alt_setting`:
```c
libusb_set_interface_alt_setting(strmh->devh->usb_devh, interface_id, 0);
if (!isochronous) {
    libusb_clear_halt(strmh->devh->usb_devh, strmh->stream_if->bEndpointAddress);
}
```

yjun123, 2023-05-25:
Thanks for your reply. Do you have any plans to submit a pull request to merge this commit into the libuvc mainline, so I don't have to use my own branch? :)

jksiezni, 2023-05-25:
I have already queued up one PR quite some time ago and still waiting for it to be merged. To me, the project is no longer maintained so it's better if you fork it and apply changes yourself. You can also use my fork if you wish.

yjun123, 2023-05-26:
Ok, thanks. I will try your suggestion.

dburckh, 2025-03-07:
I don't that it matters, but Windows actually does it in the opposite order.

[image]

drinkcat, 2026-09-05:
> I have already queued up one PR quite some time ago and still waiting for it to be merged. To me, the project is no longer maintained so it's better if you fork it and apply changes yourself. You can also use my fork if you wish.

3 years later, I'm trying to pick this up, PR would be very welcome.