libuvc/libuvc #259: Android 13

todaycodemaster, 2023-05-31:
This library is returning the 1fps on android 13 device.

Till android 12 it working well, but on android 13 is return black screen and 1fps.

Could you let me know what is the problem?

ffreality, 2023-06-06:
Did you even build "this repo" for Android ? or did you use some old android libuvc repos ?

todaycodemaster, 2023-06-06:
https://github.com/jiangdongguo/AndroidUSBCamera

In this project there is libuvc and I used it. Is it old one?

Should I upgrade with new one.

If I upgrade the libuvc it will work well? I want to know you tested on android 13 devices

ffreality, 2023-06-06:
Sorry, I thought you used library version to integrate your project. There is one but it is from 2018.

todaycodemaster, 2023-06-06:
In the new library you have never seen the errors I mentioned?

Thank for your reply. I will use new library and test again

ffreality, 2023-06-06:
I tried to create an Unreal Engine plugin. So, I wanted to build only libuvc and libusb as .so files. I can do it with libusb because it has necessary files. but I can't do it.

When I try to build project which you shared, it just creates an APK file. I don't need that. So, I didn't its performance. But I am using Samsung Galaxy S21 with Android 13 stock rom and I am using an old app from Google Play Store, I don't have any performance problem.