libuvc/libuvc #261: Can't reopen stream without reopening context on non-rooted Android

ZhivaevDmitry, 2023-07-02:
Libs:
- libusb v 1.0.26
- libuvc v 0.0.7

Device: 
- Khadas VIM3 Pro

OS: 
- Non-Rooted Android 9.0 32bit (ARMv7)

In continuation to this issue: #260 

I can't reopen stream on usb camera without recreating context. I have to re-create the context in order for the stream to start again otherwise it will not start it and freeze forever on uvc_stop_streaming. This situation is very inconvenient, because if several cameras are open in a context, you will have to reopen all of them.

Native code:

```
#include <jni.h>
#include <unistd.h>

#include <android/log.h>

#include "libusb.h"
#include "libuvc/libuvc.h"

#define PRINT_LOG_INFO(format, ...) __android_log_print(ANDROID_LOG_INFO, "!!!", format, ##__VA_ARGS__)

extern "C"
{
    JNIEXPORT jint JNICALL Java_com_example_app_MainActivity_JNIInitUvc
    (
        JNIEnv* env_,
        jclass  class_
    )
    {
        uvc_context_t* uvc_ctx = NULL;

        uvc_error_t uvc_err;

        PRINT_LOG_INFO("uvc_init");

        uvc_err = uvc_init(&uvc_ctx, NULL);

        if (uvc_err)
        {
            PRINT_LOG_INFO("uvc_init error: %s", uvc_strerror(uvc_err));
            return 0;
        }

        return (jint)uvc_ctx;
    }

    JNIEXPORT void JNICALL Java_com_example_app_MainActivity_JNIExitUvc
    (
        JNIEnv* env_,
        jclass  class_,
        jint    uvc_ctx_ptr_
    )
    {
        uvc_context_t* uvc_ctx = (uvc_context_t*)uvc_ctx_ptr_;

        PRINT_LOG_INFO("uvc_exit BEGIN");
        uvc_exit(uvc_ctx);
        PRINT_LOG_INFO("uvc_exit END");
    }

    JNIEXPORT void JNICALL Java_com_example_app_MainActivity_JNITest1
    (
        JNIEnv* env_,
        jclass  class_,
        jint    uvc_ctx_ptr_,
        jint    file_descriptor_
    )
    {
        uvc_context_t*       uvc_ctx    = (uvc_context_t*)uvc_ctx_ptr_;
        uvc_device_handle_t* uvc_devh   = NULL;

        uvc_error_t uvc_err;

        PRINT_LOG_INFO("uvc_wrap");

        uvc_err =
            uvc_wrap
            (
                file_descriptor_,
                uvc_ctx,
                &uvc_devh
            );

        if (uvc_err)
        {
            PRINT_LOG_INFO("uvc_wrap error: %s", uvc_strerror(uvc_err));
            return;
        }

        const uvc_format_desc_t *format_desc    = uvc_get_format_descs(uvc_devh);
        const uvc_frame_desc_t *frame_desc      = format_desc->frame_descs;

        enum uvc_frame_format frame_format;

        int width   = 640;
        int height  = 480;
        int fps     = 30;

        switch (format_desc->bDescriptorSubtype)
        {
            case UVC_VS_FORMAT_MJPEG:
                frame_format = UVC_COLOR_FORMAT_MJPEG;
                break;
            case UVC_VS_FORMAT_FRAME_BASED:
                frame_format = UVC_FRAME_FORMAT_H264;
                break;
            default:
                frame_format = UVC_FRAME_FORMAT_YUYV;
                break;
        }

        if (frame_desc)
        {
            width   = frame_desc->wWidth;
            height  = frame_desc->wHeight;
            fps     = 10000000 / frame_desc->dwDefaultFrameInterval;
        }

        PRINT_LOG_INFO
        (
            "First format: (%4s) %dx%d %dfps",
            format_desc->fourccFormat,
            width,
            height,
            fps
        );

        uvc_stream_ctrl_t stream_ctrl;

        uvc_err =
            uvc_get_stream_ctrl_format_size
            (
                uvc_devh,
                &stream_ctrl,
                frame_format,
                width,
                height,
                fps
            );

        if (uvc_err)
        {
            PRINT_LOG_INFO
            (
                "uvc_get_stream_ctrl_format_size error: %s",
                uvc_strerror(uvc_err)
            );

            return;
        }

        uvc_err =
            uvc_start_streaming
            (
                uvc_devh,
                &stream_ctrl,
                NULL,
                NULL,
                0
            );

        if (uvc_err)
        {
            PRINT_LOG_INFO
            (
                "uvc_start_streaming error: %s",
                uvc_strerror(uvc_err)
            );

            return;
        }

        PRINT_LOG_INFO("Sleep BEGIN");
        sleep(3); // 3 seconds
        PRINT_LOG_INFO("Sleep END");

        PRINT_LOG_INFO("uvc_stop_streaming BEGIN");
        uvc_stop_streaming(uvc_devh);
        PRINT_LOG_INFO("uvc_stop_streaming END");

        PRINT_LOG_INFO("uvc_close BEGIN");
        uvc_close(uvc_devh);
        PRINT_LOG_INFO("uvc_close END");
    }
}

````

Java code variant 1:
````
    ...
    
    private void handleUsbDeviceCanBeOpened()
    {
        UsbManager usbManager =
            (UsbManager)getApplicationContext().getSystemService(Context.USB_SERVICE);

        int uvcContextPtr = JNIInitUvc(); // init context one time

        for (int iterationIndex = 0; iterationIndex < 1000; ++iterationIndex)
        {
            Log.d("!!!", "-----Iteration " + iterationIndex+ " BEGIN");

            UsbDeviceConnection usbDeviceConnection = usbManager.openDevice(mUsbDevice);
            {
                JNITest1(uvcContextPtr, usbDeviceConnection.getFileDescriptor());
            }
            usbDeviceConnection.close();

            Log.d("!!!", "-----Iteration END");
        }

        JNIExitUvc(uvcContextPtr); // exit context one time
    }
    
    private static native int   JNIInitUvc();
    private static native void  JNIExitUvc(int uvc_ctx_ptr_);
    private static native void  JNITest1(int uvc_ctx_ptr_, int file_descriptor_);
    
    private UsbDevice mUsbDevice;
    
    ...
````
Output variant 1:

> I/!!!: uvc_init
> D/!!!: -----Iteration 0 BEGIN
> I/!!!: uvc_wrap
> I/com.example.app: type=1400 audit(0.0:665): avc: denied { read } for name="usb" dev="tmpfs" ino=4524 scontext=u:r:untrusted_app:s0:c59,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
> I/com.example.app: type=1400 audit(0.0:666): avc: denied { open } for path="/dev/bus/usb" dev="tmpfs" ino=4524 scontext=u:r:untrusted_app:s0:c59,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
> I/!!!: First format: (YUY2) 640x480 30fps
> I/com.example.app: type=1400 audit(0.0:668): avc: denied { open } for path="/sys/bus/usb/devices" dev="sysfs" ino=9774 scontext=u:r:untrusted_app:s0:c59,c256,c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
> I/!!!: Sleep BEGIN
> I/!!!: Sleep END
>     uvc_stop_streaming BEGIN
> I/!!!: uvc_stop_streaming END
>     uvc_close BEGIN
> I/!!!: uvc_close END
> D/UsbDeviceConnectionJNI: close
> D/!!!: -----Iteration END
>     -----Iteration 1 BEGIN
> I/!!!: uvc_wrap
> I/!!!: First format: (YUY2) 640x480 30fps _**It writes that it has launched a stream, but in fact there are no frames.**_
> I/!!!: Sleep BEGIN
> I/!!!: Sleep END
> I/!!!: uvc_stop_streaming BEGIN
> _**At this point the app freezes forever.**_

Java code variant 2:
````
    ...
    
    private void handleUsbDeviceCanBeOpened()
    {
        UsbManager usbManager =
            (UsbManager)getApplicationContext().getSystemService(Context.USB_SERVICE);

        for (int iterationIndex = 0; iterationIndex < 1000; ++iterationIndex)
        {
            Log.d("!!!", "-----Iteration " + iterationIndex+ " BEGIN");

            int uvcContextPtr = JNIInitUvc(); // init context every iteration

            UsbDeviceConnection usbDeviceConnection = usbManager.openDevice(mUsbDevice);
            {
                JNITest1(uvcContextPtr, usbDeviceConnection.getFileDescriptor());
            }
            usbDeviceConnection.close();

            JNIExitUvc(uvcContextPtr); // exit context every iteration

            Log.d("!!!", "-----Iteration END");
        }
    }
    
    private static native int   JNIInitUvc();
    private static native void  JNIExitUvc(int uvc_ctx_ptr_);
    private static native void  JNITest1(int uvc_ctx_ptr_, int file_descriptor_);
    
    private UsbDevice mUsbDevice;
    
    ...
````
Output variant 2:
> _**It will work correctly for 1000 iterations and there will be valid frames in each iteration**_

FrancoisDn, 2023-09-12:
Have you find a way to solved your problem without having a new context each time without recreating the context?
I have the exact same problem on Android, but also on Linux

ZhivaevDmitry, 2023-09-12:
> Have you find a way to solved your problem without having a new context each time without recreating the context? I have the exact same problem on Android, but also on Linux

Unfortunately no, I am still eagerly awaiting a response from the gurus who implemented all of this.

FrancoisDn, 2023-09-12:
I test with this commit https://github.com/damijans/libuvc/commit/0480813669cc54eca18c25177e1b849c8abc433e from PR https://github.com/libuvc/libuvc/pull/224 (cf. issue https://github.com/libuvc/libuvc/issues/227) and the problem is solved on my side. On Linux and Android

zongren, 2023-09-18:
@ZhivaevDmitry any demo repo in github?