libuvc/libuvc #280: preview of uvc camera is flickering by Android

yymjr, 2024-05-14:
I have two uvc camera devices. The preview of one device is invalid and the other is flickering when I use libusb and libuvc code of master brach by Android.

The sample code can run as Ubuntu and Linux, or use code from [Android UVC Camera](https://github.com/jiangdongguo/AndroidUSBCamera) and it's work fine.

run log file
[log.txt](https://github.com/libuvc/libuvc/files/15301959/log.txt)

yymjr, 2024-05-14:
I found that the main reason is that preview data is received is incomplete. The received data length is 590172 When running on Ubuntu, but it's changes at any times on Android.

`void UVCCamera::frame_callback(uvc_frame_t *frame, void *) {

    LOGE("hecsion callback! length = %zu,\n", frame->data_bytes);
    JNIEnv *env{nullptr};
    JavaVM *vm = JVM::GetInstance()->jvm();
    if (vm == nullptr) {
        return;
    }
    int result = vm->AttachCurrentThread(&env, nullptr);
    if (result != JNI_OK) {
        return;
    }

    env->CallVoidMethod(
            object_camera_help_ref, method_frame_callback, (jlong) frame->data,
            (jlong) frame->data_bytes, (jint) frame->width, (jint) frame->height);
    if (env != nullptr) {
        vm->DetachCurrentThread();
        env = nullptr;
    }
}`

tomjeans, 2024-06-27:
using 0.0.7 ver

yymjr, 2024-06-27:
> using 0.0.7 ver

what changes between 0.0.7 version and the master branch caused the problem?

tomjeans, 2024-10-30:
or change the LIBUVC_NUM_TRANSFER_BUFS

yymjr, 2024-10-30:
> or change the LIBUVC_NUM_TRANSFER_BUFS

your answer is correct,because when I adjusted the LIBUVC_NUM_TRANSFER_BUFS is 10 it is worked fine.
Although I have solved the probelm,I still appreciate your reply.

closed 2024-10-30 by yymjr