libuvc/libuvc #282: _uvc_stream_params_negotiated should skip dwMaxPayloadTransferSize check

fbzhong, 2024-07-30:
I have a usb camera and the _uvc_stream_params_negotiated always return FALSE. According to the log:
```
14:17:41.083  D  [11098*stream.c:604/_uvc_stream_params_negotiated] Negotiation failed: required format 1, frame 3, size 800; actual format 1, frame 3, size 3072
```
The reason is required.dwMaxPayloadTransferSize is smaller than actual.dwMaxPayloadTransferSize.

According to the #273 , I think the dwMaxPayloadTransferSize should be ignored.

soonwonjang, 2025-02-13:
I agree with you. I experienced the same issue with the Kiyo Pro Ultra and Logitech MX Brio

fdegaulejac, 2025-07-02:
Same with my HD USB Video Capture Card from UGreen, works when removing the check on dwMaxPayloadTransferSize like suggested

drinkcat, 2026-09-04:
(updated)

I'm not fully clear which of the 2 approaches in #277 or #292 is better (I'd lean towards an updated #277: https://github.com/drinkcat/libuvc/pull/17, but I'm not sure if that fixes the issue).

drinkcat, 2026-11-14:
Would be good to retest this on a recent libuvc, I'm not 100% clear if #292 is enough (or if we need something like #320).