libuvc/libuvc #283: Get crash when USB interface gone(unmounted from pc) at uvc_close

msarafati, 2024-08-02:
my code will crash at line uvc_close(devh);
when client remove the camera from PC, the code will call uvc_close(devh) , but It gets:
camera_project: ../../libusb/os/threads_posix.h:58: usbi_mutex_destroy: Assertion `pthread_mutex_destroy(mutex) == 0' failed.

yangyagami, 2024-08-29:
Update your libusb.

yangyagami, 2024-08-29:
https://github.com/libuvc/libuvc/issues/url

> my code will crash at line uvc_close(devh); when client remove the camera from PC, the code will call uvc_close(devh) , but It gets: camera_project: ../../libusb/os/threads_posix.h:58: usbi_mutex_destroy: Assertion `pthread_mutex_destroy(mutex) == 0' failed.

yangyagami, 2024-08-29:
> my code will crash at line uvc_close(devh); when client remove the camera from PC, the code will call uvc_close(devh) , but It gets: camera_project: ../../libusb/os/threads_posix.h:58: usbi_mutex_destroy: Assertion `pthread_mutex_destroy(mutex) == 0' failed.

[https://github.com/libusb/libusb/issues/1199](https://github.com/libusb/libusb/issues/1199)

msarafati, 2024-08-29:
thanks, the issue is solved, i used:
V1.0.27 for libusb
V0.0.7 for libuvc 
and it works

closed 2024-08-29 by msarafati