libuvc/libuvc #284: Project status?

qwertychouskie, 2024-10-01:
It seems progress on libuvc has stalled, despite many open PRs ranging from simple bugfixes to bigger improvements.  If the current maintainers don't currently have enough time to work on the project, I would suggest putting out a call for new maintainers to step up and help work through the PR backlog and get a new release out.

drinkcat, 2026-09-03:
@mcuee I'm wondering if this got anywhere, and if you managed to reach to the current maintainers somehow.

I started having a look at the currently open PR (https://github.com/drinkcat/libuvc/issues/1), and many of them are relatively low-hanging fruits, that I merged into my fork (https://github.com/drinkcat/libuvc/tree/master), sometimes with minor (or more major) modifications.

That fork should be reasonable usable. I'll still try to find other PR (and random commits on various forks), and it'd be good to improve CI coverage.

I don't particularly want to maintain a permanent fork though. I would feel better with a second set of human eyes on the MR I merged (Claude is helpful but.... doesn't really count IMHO ,-P), and more testing is needed for the HW-specific quirks. I think that can only be done well as part of the original project.

Anyhow, I'd be happy to help maintain libuvc, at least for a while to help clear the current backlog (this project generally feels close to "done" and just needs a few touchups...)

Thanks,

mcuee, 2026-09-03:
@drinkcat 

I have not heard anything from the current maintainers.

Thanks for your efforts!

tore-espressif, 2026-09-03:
From my side, I am not and never have been maintainer of this project. I just implemented few quirks for our embedded ESP32 series of chips. 

But eventually, we moved from libuvc to our custom tailored solution in https://github.com/espressif/esp-usb/tree/master/host/class/uvc/usb_host_uvc

It is not perfect and compared to libuvc it lacks handling of _many_ non-conforming USB cameras that are in the field. But at least we understand what is going on inside :)

mcuee, 2026-09-03:
> From my side, I am not and never have been maintainer of this project. I just implemented few quirks for our embedded ESP32 series of chips.

I see. You have the code commit right though. Just wondering if you have the right to add collaborators/admins.

mcuee, 2026-09-03:
FYI, I have code commit right to libusbdotnet github repo even though I am not the admin and I cannot add collaborators/admins.

I am kind of able to maintain libusbdotnet project as of now even with some limitations.
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/276

ktossell, 2026-09-03:
Hi! Author and original maintainer here. I handed off maintenance for this project but I think the other maintainer hasn’t had time to attend to the project in some time. I appreciate the work you’ve done and I’ll add you to the project so you can go ahead and merge your improvements.

drinkcat, 2026-09-03:
Thanks @ktossell , appreciated! (and thanks for libuvc, it still seems quite popular!)

waddlesplash, 2026-09-03:
> I handed off maintenance for this project but I think the other maintainer hasn’t had time to attend to the project in some time. 

I am generally busy with other things, yes, but also when you handed off maintenance, I think you set things up so that I can't merge my own changes, but only ones reviewed by others? So then I couldn't really do much and so dropped working on it.

mcuee, 2026-09-04:
@ktossell 

Thanks a lot for granting me and @drinkcat code commit rights.

drinkcat, 2026-09-05:
Closing this for now, the next obvious step should be to merge simple fixes, and make a new release (#310), any help appreciated.

closed 2026-09-05 by drinkcat