libuvc/libuvc #286: integrated webcam (ACER SP514-51N) not working

crysman, 2024-11-04:
Hello,
I don't know if this is the right place (in case it's not, please redirect me to more appropriate site), but trying - the integrated camera on ACER SP514-51N is not working in GNU/Linux. On the same machine, when MS Windows is booted, works just fine.

According to ACER forum (link below), it is a CHICONY model "**_CJFKF28_**"

**How to make the camera working?**

Here is some debug info from GNU/Linux (latest stable POP_OS, currently on kernel `6.9.3-76060903-generic #202405300957~1726766035~22.04~4092a0e`):
```
spin5 :: ~ » sudo lsmod | grep -iE "video|cam"
videobuf2_vmalloc      20480  0
videobuf2_memops       16384  1 videobuf2_vmalloc
videobuf2_v4l2         36864  0
videobuf2_common       81920  3 videobuf2_vmalloc,videobuf2_v4l2,videobuf2_memops
videodev              356352  4 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov2740
mc                     81920  5 v4l2_async,videodev,videobuf2_v4l2,videobuf2_common,ov2740
video                  73728  3 acer_wmi,xe,i915
wmi                    28672  3 video,acer_wmi,wmi_bmof
spin5 :: ~ » sudo dmesg | grep -iE "video|cam"
[    0.000000]   AMD AuthenticAMD
[    0.100395] PCI: ECAM [mem 0xc0000000-0xcfffffff] (base 0xc0000000) for domain 0000 [bus 00-ff]
[    0.100395] PCI: not using ECAM ([mem 0xc0000000-0xcfffffff] not reserved)
[    0.224013] PCI: ECAM [mem 0xc0000000-0xcfffffff] (base 0xc0000000) for domain 0000 [bus 00-ff]
[    0.226296] PCI: ECAM [mem 0xc0000000-0xcfffffff] reserved as ACPI motherboard resource
[    0.495588] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    7.005562] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[    7.006168] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input18
[    8.942344] videodev: Linux video capture interface: v2.00
[ 1755.195885] usbcore: registered new interface driver uvcvideo
[ 2471.824739] usbcore: deregistering interface driver uvcvideo
spin5 :: ~ » sudo lsusb                       
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 04f3:0c7f Elan Microelectronics Corp. ELAN:Fingerprint
Bus 003 Device 004: ID 8087:0033 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
spin5 :: ~ »
```

Here is the info from MS windows device list:
```
Intel(R) ADL AVStream Camera DISPLAY\INT3480\4&2DED5861&0&UID144512
VIDEO\VEN_8086&DEV_46A6&SUBSYS_16131025&REV_0C&INT3480
VIDEO\VEN_8086&DEV_46A6&SUBSYS_16131025&INT3480 VIDEO\VEN_8086&DEV_46A6&INT3480 VIDEO\INT3480
INTC#ACPI3480 VIDEO\INT3480 oem81.inf:VIDEO\INT3480,iacamera64.NT,12/30/2021,63.22000.3.6744 _SB.PC00.GFX0.IPUA
ACER FHD User Facing
```

Thanks a lot!

Related:
https://community.acer.com/en/discussion/comment/1302958#Comment_1302958

drinkcat, 2026-09-04:
Not a libuvc issue, probably a Linux kernel issue. It might just work fine 2 years later...

closed 2026-09-04 by drinkcat

BiNeurale, 2026-09-04:
If your camera light blinks on and off while the picture stays black, and stops blinking when the window loses focus, this may be it — and it is not a driver or a privacy setting.
Acer installs a Device MFT (AcerMediaService.dll) that sits inside the Windows Media Foundation camera pipeline and processes every frame. After a recent Windows 11 update it loads but stops delivering frames, so every app gets black while Device Manager still reports the camera as healthy. The blinking is the app opening the stream, getting nothing, timing out and retrying.
The reason the registry fixes in this thread stop working after a reboot: the service AcerARTAIMMXDriverService re-registers the plug-in at every startup through DetectCameraDMFT.exe. You have to stop that service too.
The neat part is that Acer already ships the uninstaller for it — UninstallAcerCameraDMFT.exe, inside its own driver package.
I wrote it up with a script that runs read-only by default and tells you whether this is actually your problem before touching anything: https://github.com/BiNeurale/acer-camera-black-screen-fix
If the read-only run does not print CameraDeviceMftClsidChain, this is not your fault — stop there, the same symptom has several other causes.