libuvc/libuvc #290: Bug in stream.h, _uvc_stream_callback function, resubmit flag

JohnAtBeach, 2025-01-25:
In stream,c function _uvc_stream_callback, I believe there is a bug. 

When 
transfer->status = LIBUSB_TRANSFER_COMPLETED 
and transfer->num_iso_packets =0, the resubmit flag should be cleared after processing the payload.

This seems to cause faults when occuring during streaming iso_transfers.

darktiny, 2025-02-03:
May I ask in what situation this would happen?

JohnAtBeach, 2025-02-03:
It probably wouldn't occur (unless intentional).  In reviewing the Android Studio logcat during troubleshooting of an android_usbfs.c file, what appeared to be duplicate bulk transfer was resubmitted after a ISO streaming session (strmh->running = 1) had begun.  However, the bulk transfer entry was likely delayed logging of a earlier transfer used for device negotiation.  

If using bulk transfers for streaming instead of iso, you probably would want to keep resubmitting completed transfers whenever the running flag is set.

closed 2025-02-03 by JohnAtBeach